Ultimaker/Cura #15620: errore nella visualizzazione del piano di stampa

Teurus92, 2023-05-28:
### Application Version

5.3.1

### Operating System

window 10

### Printer

sovol sv06

### Reproduction steps

io possiedo una sovol sv06 plus. su cura manca la stampante quindi ho selezionato sia il profilo della sovol sv 06 che della prusa mk3s. in entrambi i casi ho modificato il piano della stampante impostando x=300 ; y=300 ; z=340. quando vado a dare l'ok il piano che mostra e 300x300 ma l'asse z risulta piu piccolo di 340
[image]

### Actual results

come si evince dalla foto allegata il piano non risulta corretto e risulta più basso del dovuto. quindi un piano che dovrebbe essere 300x300x340 risulta di 300x300x250
[image]

### Expected results

dovrei visualizzare il piano correttamente modificato

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

vorrei sapere come aggiustare il bug

GregValiant, 2023-05-28:
Go the "Settings/Manage Printers" and then "Machine Settings".
Change the "Z(height)" to 340.  Let us know if that works for you.
This is from the definition file for the Sovol sv06:
"machine_height": { "default_value": 250 }

So there is no bug here, just a setting issue and I'll remove the bug label.  I hope you understand.

Teurus92, 2023-05-28:
Se leggi il bug sta nel fatto che modificando la z in 340 non lo modifica

Inviato da Outlook per Android<https://aka.ms/AAb9ysg>

Closed #15620<https://github.com/Ultimaker/Cura/issues/15620> as completed.

GregValiant, 2023-05-28:
OK.  Maybe it was Google Translate.
I've added the bug label back.
I can't duplicate this.  When I set the printer to 300x300x340 it stays there.
There have been other problems with the Machine Settings dialog box though and so I will put the bug label back on.

If you open the printer definition file in a text editor (Windows Notepad) and change the line to:
"machine_height": { "default_value": **340** }
You should get the full build volume when you restart Cura.
The file is in "C:\Program Files\UltiMaker Cura 5.3.1\share\cura\resources\definitions" and is named "sovol_sv06.def.json".