Ultimaker/Cura #15621: After slicing models, the highlighted red area for supports is mislocated

GoguyT3D, 2023-05-28:
### Application Version

UltiMaker Cura 5.4.0-alpha+cura_10389_148

### Operating System

Windows 11

### Printer

Ultimaker S3

### Reproduction steps

After slicing my models which were directly imported from Fusion 360 I noticed while browsing the "Preview" section to check my z seam was residing where I wanted it, I went back to "prepare" and I noticed something a little different from other versions of cura. The area highlighted in red (representing need for supports) almost seems to be on the opposite side of the model (rather than the bottom). This does not effect the supports that were created right after.
[image]
[image]

### Actual results

No serious problems occur although I have used the "red section" in the past to see if my model needed or maybe even required supports. I don't know if this is intentional because its the nightly build, or if there is a bug.
Rundown:
Model slices normally
Supports are rendered normally via the angle selected
In "Prepare" section of viewing the models the highlighted red section is not where it should be
This is not a detrimental bug although it is noticed as an error.

### Expected results

While looking at the "Prepare" section of the model, the highlighted red section should be displayed on the parts of the model that are <(user defined)-degrees that are chosen for supports in printing profile, and should be mostly viewable from under the print bed rather than above.
[78.26% Oil-Holder x 4.zip](https://github.com/Ultimaker/Cura/files/11584359/78.26.Oil-Holder.x.4.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/11584370/cura.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

There is no SERIOUS problem though it is something that needs to be considered before the next release.

GregValiant, 2023-05-28:
Thanks for the report.
When I first looked I thought I would find model errors but I think you are right.  Cura is calling for "Anti-Gravity Supports".
Adjusting the "Support Overhang Angle" changes the size of the red areas.  Flipping the part over has no effect as the red areas stay on top.  (I wonder if someone missed a negative sign.)
We'll see if @MariMakes has some insight on this.

GoguyT3D, 2023-05-28:
@GregValiant Let me open it back up, and see if:
Exporting from Cura then importing
Exporting from Fusion360 then importing
Importing directly from Fusion360
or reopening the 3mf in Cura causes this

GoguyT3D, 2023-05-28:
@GregValiant This is wierd, bear with me:
-Exporting 1 of the models from cura as a binary STL, clearing the buildplate and reimporting the same stl I just exported causes the same problem, red on top
-Reopening the original 3mf in cura again causes the same problem, red on top
-Clearing the buildplate then importing directly from Fusion360, shows no red at all until supports are selected and modified to 30 degrees (sorry my keyboard doesnt have the degree symbol) then the support rendering is correct, red at the >30 degree angle on the bottom of model. Picture included
[image]

GoguyT3D, 2023-05-28:
Now I am going to try to replicate this, then resave the 3mf overwriting the original to see if: Reopening the 3mf has normal support redness to begin with

GoguyT3D, 2023-05-28:
Yes, after my previous step the file reacts normally and the red is on the bottom angles now....Still kina confused as to why this happened in the first place I was almost about to say I thought it had to do with the filename that I chose for the 3mf (78.26% Oil-Holder x 4)but I just proved that to be false because it loads fine now.
[image]

GregValiant, 2023-05-28:
I'm going to stick with my original opinion that I'm clueless.  I don't think it's Cura though.  Enough people use Fusion that I think this would have come up before.  Nevertheless, I'll leave the bug label on.  The Cura team has different tools and maybe they will come up with a cause.

I do know this...
ALT + 0176 is degree symbol °
ALT + 0179 is cubed symbol ³
ALT + 0178 is squared symbol ²

GoguyT3D, 2023-05-28:
° great thanks @GregValiant