Ultimaker/Cura #15622: 5.3.1 tree supports seem to print starting in the air

jayyyin, 2023-05-28:
### Application Version

5.3.1

### Operating System

windows 10

### Printer

Ender-3

### Reproduction steps

1. enable tree support with the same settings as I had in older versions
2. only changes from default are:
  support overhang angle is 55
  support density is 5%
  support top distance 0.0 (if this isn't set the support doesn't connect with the structure its trying to support)

### Actual results

the printer starts the floor of the tree support almost 1cm off the print bed resulting in tangled mess

### Expected results

floor layer is the same as any other

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3E3V2_frame_top.zip](https://github.com/Ultimaker/Cura/files/11584524/CE3E3V2_frame_top.zip)

GregValiant, 2023-05-28:
Thanks for the report and the file.
This bug occurs when you call for "Enable Support Brim" and have the Build Plate Adhesion set to "None".
If you set the build plate adhesion to Skirt or Brim the problem will go away.
I'll mark this as a duplicate and the Cura team will take a look.

jayyyin, 2023-05-28:
Ah I see, would disabling support brim fix this?

GregValiant, 2023-05-28:
Yes, either way.  The problem occurs when Support Brim is enabled and Build Plate Adhesion is set to "None".
If they are both off it's OK and if they are both turned on it's OK.