Ultimaker/Cura #15623: Bed adhesion // initial layer height // lower for walls only

leo-na-rd, 2023-05-28:
### Is your feature request related to a problem?

bed adhesion

### Describe the solution you'd like

adjust the height of the nozzle slightly lower (ex: 0.05mm) when printing the walls only on the initial layer

### Describe alternatives you've considered

adjusting it manually on the fly

### Affected users and/or printers

absolutely everyone

GregValiant, 2023-05-29:
I don't understand.  Is this a fix for a build plate adhesion problem?
I run the Initial Layer Flow at 105% for PLA.  That works for me.

There would be no advantage to dropping the layer height if the flow volume was based on that new layer height.
When you manually lower the height (25%?) then the flow was calculated for the regular layer height and you are in effect over-extruding the walls.  Is that what you want?
It would be easy to post-process by increasing the "Initial Layer Wall Flow" by some amount by adding M221 S??? for the walls and then returning to M221 S100 for the skins.  I really don't see any reason to adjust the layer height if for no other reason than it would cause the following layer walls to be under-extruded.

leo-na-rd, 2023-05-29:
hi @GregValiant , thank you for your reply.

Let's start by with what we can easily agree:
1) lower nozzle height increases bed adhesion
2) when the nozzle height is too low, it squishes the surface, decreases the quality, creates blops, etc.

The solution you're proposing (105% flow on the initial layer) is in my opinion, a partially solution. 
It will definitely help with the adhesion, but it will also introduce a new problem with surface quality.

After 3 years of thinkering, I have experience great success by manually decreasing z (babystep z feature on the firmware) when the print starts (printing walls), then I increase the Z height back to original value when it starts drawing the surface (bottom pattern). Once the walls are printed, the rest of the layer printing (surface) goes smooth.

So in conclusion, I believe we should have an option in the **walls** settings, to set a negative z offset for the initial layer.

GregValiant, 2023-05-29:
Thanks for that.  It clears up a bit of the confusion I was having.

GregValiant, 2023-05-29:
I think this Post-Processor will do.  Let me know.
Unzip and put in the "C:\Program Files\UltiMaker Cura 5.3.1\share\cura\plugins\PostProcessingPlugin\scripts" folder or whatever is the equivalent on Linux or Mac.
[InitLayerZWalls.zip](https://github.com/Ultimaker/Cura/files/11593667/InitLayerZWalls.zip)

cnaslain, 2025-01-25:
This is exactly what I was looking for. I'm gonna testing it.
If it works well; would it be possible to integrate this into the official plugin?

GregValiant, 2025-01-25:
Frankly, I'd forgotten about that script.
I will not submit it to Cura because I very much doubt if it would be accepted.  There are a few reasons:

- "Initial Layer Flow" is sufficient for this.  Trying to increase the "Bed Adhesion" by moving the Z up and down is really a niche fix.  There is a very small percentage (near 0%) of users who would want such a thing.
- It automatically introduces elephant foot because all it really does is make the "Initial Layer Wall Line Width" wider.  It's a flow thing.
-- Example: Your line width is 0.4 and your Initial Layer Height is 0.20.  If you drop the Z for the walls from 0.20 to 0.15, and you do not adjust the flow rate, then all you are really doing is over-extruding the Initial Layer Walls and that extra plastic has to go somewhere and it can't go taller so it goes wider.  It pushes the effective Line Width to 0.50.  That's "elephant foot" by definition.
- That particular script leaves the second layer walls under-extruded because the second layer Z is not adjusted and neither is the flow, but the effective layer height of the 2nd layer walls is 0.25 _but_ the flow was calculated at a layer height of 0.20 and so the second layer walls will be under-extruded by 25%.
- The "Initial Layer Skin Overlap" will increase and cause over-extrusion where the skin intersects the walls because the walls are wider than what was planned on.
- As written - that script does not take into account "One at a Time" print sequence.  That may seem like a small thing, but someone would complain.

This version of the script does adjust the second layer flow to account for the thicker "Wall Layer Height" and it is compatible with "One at a Time" print sequence.  I still don't like the idea of doing it this way.  If the Bed Adhesion is a problem it can be fixed using conventional means rather than doing it _(IMHO a dumb way)_ like this.

[InitLayerZWallsV2.zip](https://github.com/user-attachments/files/18546404/InitLayerZWallsV2.zip)