Ultimaker/Cura #15624: cura gave an error when slicing

apacheGitHub, 2023-05-29:
### Cura Version

5.3.1

### Operating System

MacOS Monterey 12.6.6

### Printer

Artillery sidewinder x2

### Name abnormal settings

no these settings worked before

### Describe model location

i re tried it and it worked

### Describe your model

i had sliced it before, i then changed the support to Tree

### Add your .zip here ⬇️

[ASX1_cowl joined.3mf.zip](https://github.com/Ultimaker/Cura/files/11590384/ASX1_cowl.joined.3mf.zip)

apacheGitHub, 2023-05-29:
on re trying it , it worked, so not sure why it failed

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic