Ultimaker/Cura #15627: Slicing failed With Tree Support

dvdaniels1967, 2023-05-30:
### Cura Version

5.3.1

### Operating System

Windows 10

### Printer

Anycubic Chiron

### Name abnormal settings

Support Structure, Support Placement

### Describe model location

Center Buildplate

### Describe your model

An earlier revision sliced. This is a larger version

### Add your .zip here ⬇️

Just created this model to print a prototype. Using tree support as the support structure and touching buildplate as the support placement, it will not slice with the default Tree Support Branch Angle of 40. If you change it to 50 it works.

[AC_RainBarrelOverflow (1).zip](https://github.com/Ultimaker/Cura/files/11595237/AC_RainBarrelOverflow.1.zip)

GregValiant, 2023-05-30:
Thanks for the report and the file.
The file wouldn't slice for me either.  Rather than changing any settings I moved the model away from the build plate midpoint and it sliced.
This appears to be a duplicate of several other reports and I'll mark it as a dupe.  The Cura team will take a look.

On thing I noticed is that the tube spigot doesn't go through to the cavity.  (It might be your design intent.)
[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic