Ultimaker/Cura #15635: Shiloh's Sugago Keychain

Shilohami, 2023-05-30:
### Cura Version

5.3.1

### Operating System

Windows 10

### Printer

Ender 3

### Name abnormal settings

No abnormal settings

### Describe model location

Laid flat on the back of the keychain, in the center of the build plate. No other positions slice either.

### Describe your model

A keychain I've already sliced in Chitubox, and printed over 500 on an Elegoo Saturn.

### Add your .zip here ⬇️

[sugagokeychainfailedcuraslice.zip](https://github.com/Ultimaker/Cura/files/11607162/sugagokeychainfailedcuraslice.zip)
[sugagokeychainfailedcuraslice.zip](https://github.com/Ultimaker/Cura/files/11607172/sugagokeychainfailedcuraslice.zip)

GregValiant, 2023-05-31:
Thanks for the report and the files.
This is a well known bug.  The workarounds are to:

- move the model away from the Cura build plate midpoint
- rotate the model
- change the Initial Layer Height.

I'll mark this as a duplicate.  I'm going to close it as well since this bug has been reported numerous times and hopefully it will be fixed in 5.4.  I hope you understand

I moved the model a little bit and it sliced.  This is with a 0.2 nozzle since the "2023" didn't want to print with a .4 or a .3.
[image]

closed 2023-05-31 by GregValiant

Shilohami, 2023-06-05:
Thank you very much, I greatly appreciate your assistance!