Ultimaker/Cura #15645: "No profiles matching configuration of this extruder" with TPU 95A

mobilePilot, 2023-05-31:
### Application Version

5.1.1

### Operating System

Windows 10

### Printer

Voron 0.1

### Reproduction steps

1. Open Cura, drag-and-drop an STL model
2. Click on the Material drop-down menu (tab hovering in the middle Upper part of visualization space)
3. Select a (default) Custom Material, specifically "TPU 95A" or "PVA" or "BVOH"

### Actual results

i. 'Material' icon in that "selection tab" displays a yellow exclamation mark. 
ii. Hover the mouse to read the message ""No profiles matching configuration of this extruder"
iii. 'Profile' drop-down menu shows "Not Supported" (under "Print settings" upper-right corner box)

### Expected results

i. No exclamation mark as other materials such as PLA, ABS, PETG
ii. No warning message
iii. 'Profile' drop-down menu shows, for example, "Fast - 0.2mm", or "Normal - 0.15mm", etc.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Note I am selecting the Generic materials, which came as defaults in Cura and cannot be edited. This issue also happens if I create a new material and write the 'Material Type' as "TPU 95A" or any of the other problematic materials mentioned above.

GregValiant, 2023-06-01:
Thanks for the report.
I think you might be missing a couple of files for that printer.  There should be Quality files for the materials and nozzle size you are using.   I don't see a printer definition file for a "Voron 0.1".
This is probably a Voron problem as UltiMaker doesn't create the files for 3rd party printers.  Have you tried altering a Voron2 Quality file or checking Facebook or Reddit to see if someone else has run into the problem?

mobilePilot, 2023-06-02:
Thanks for that quick reply! I am not 100% I understand your response, so allow me to clarify some points:

1.1) Voron states that "Cura comes with built-in profiles for most Voron printers"
1.2) The profile exact name is "Voron 0" under "VoronDesign" brand, which was indeed pre-built inside Cura. 
2) I can give a shot to change/alter/create this/these 'Quality file' you refer, where can I find the existing files so to understand how they work? I didn't find them in "Program Files/Ultimaker Cura" folder 
Appreciate your help

Superbrain8, 2023-06-09:
there is no quality settings defined within cura to allow the use of TPU with any of the Vorons

GregValiant, 2023-06-09:
_"1) Voron states that "Cura comes with built-in profiles for most Voron printers"_
That appears to be true for the Voron0.

It's the Quality file that is the problem.  Voron didn't provide a quality file for TPU.  That doesn't mean you can't print with it.  You can use any profile and change the Temperature, Speed, and (probably) Retraction settings and then save the Setting Profile so you can use it any time.
If you start out by making a Custom Material based on Generic TPU you might even get rid of that yellow box.

Superbrain8, 2023-06-10:
making on based on generic TPU wont work, since the profile for the Vorons dont have tpu defined at all.
Making a Material profile based on PLA would work tho, but its not the clean solution.
Could write the needed Configs, but i have no way to verify them.

> It's the Quality file that is the problem. Voron didn't provide a quality file for TPU. That doesn't mean you can't print with it. You can use any profile and change the Temperature, Speed, and [...]