Ultimaker/Cura #15649: Slicing faile on Z(V5 project

j-snijder, 2023-06-01:
### Cura Version

5.3.1

### Operating System

Windows 11

### Printer

ZOnestar Z9V5

### Name abnormal settings

as far as I know default

### Describe model location

mid center on plate

### Describe your model

4 cilinders for testing, sliced before on Prusa

### Add your .zip here ⬇️

[files.zip](https://github.com/Ultimaker/Cura/files/11624530/files.zip)

GregValiant, 2023-06-01:
Thanks for the report.
This sliced for me on the first try.  There have been several reports of a bug that will keep models from slicing if they are at the bed midpoint.,  You can try moving the group of models.  Something rotating or changing the Initial Layer Height can fool this bug.

I'll mark this as a duplicate slicing error.  The Cura team has a lot of examples of this behavior and hopefully it will be fixed in 5.4.  I'll go ahead and close this.  I hope you understand.

closed 2023-06-01 by GregValiant

j-snijder, 2023-06-05:
Nope, slicing fails again now when slicing a simple coin on E2, moved it around the bed and on 10, read TEN places slicing fails.

Somehow you all manage to get more bugs in every time Cura is changed :(

GregValiant, 2023-06-05:
Make no mistake, this bug is a PITA and has been reported a LOT.  One of the oddities of it is that my system rarely suffers from it.  I know from reading the reports here that some installations barely function and that includes when UM printers are involved.
I keep 4.13.1 handy just in case but for 99% of my stuff I use 5.3.1.

j-snijder, 2023-06-06:
Yeah, the good old days 😒Now I am convicted to Prusa, and believe me, thats not even close to Cura.I want my Cura back 😭

j-snijder, 2023-06-06:
Btw, it started when the mmu printer(s) from creality were added I believe to remember

GregValiant, 2023-06-06:
What is an "mmu" printer?
I open a lot of 3mf files and I probably have had every printer in the list installed at one time or another.  My own printer is an Ender 3 Pro and as I say; this "model slicing error" bug hasn't been an issue for me.

j-snijder, 2023-06-06:
Mmu ? Have no idea 🤣Color printers or multi extruder printers like Zonestar Z8 or Z9 are the only ones I am targetting in my question.Its a multi extruder/color thing I guess

j-snijder, 2023-06-06:
https://www.google.nl/search?q=mmu+prusa&sxsrf=APwXEddI_q5q4jru6ac3kgqNQeA8OwbaqQ%3A1686055992684&source=hp&ei=OCx_ZJaDJ_2X9u8P3KqC2A4&oq=mmu&gs_lcp=ChFtb2JpbGUtZ3dzLXdpei1ocBABGAYyBQgAEIAEMgUIABCABDIFCAAQgAQyBQgAEIAEMgUIABCABDIFCAAQgAQyBQgAEIAEMgcIABCABBAKOgYIswEQhQQ6CwgAEIAEELEDEIMBOgUILhCABDoLCC4QgAQQsQMQgwE6EQguEIAEELEDEIMBEMcBENEDOgsILhCABBDHARCvAToICAAQgAQQsQNQnQhY-gpgrS1oAXAAeACAAZ8BiAHWApIBAzEuMpgBAKABAbABAQ&sclient=mobile-gws-wiz-hp

GregValiant, 2023-06-06:
OK.. Multi-Extruder.