Ultimaker/Cura #15654: Slicing fails when attempting "Wire Printing"

SunnyJames, 2023-06-02:
### Cura Version

5.3.1

### Operating System

MacOS Ventura

### Printer

UM3

### Name abnormal settings

Wire Printing

### Describe model location

Center of the build plate, orientation does not matter

### Describe your model

Simple step design to test plastic transparency.  This is not a mesh issue. Printing slices normally when "Wire Printing" is not enabled.  This feature causes slicing to fail

### Add your .zip here ⬇️

[Archive.zip](https://github.com/Ultimaker/Cura/files/11631537/Archive.zip)

GregValiant, 2023-06-02:
Thanks for the report.
Yes, Wire Printing is broken in 5.3.1.  From what I've seen and read I don't think it's scheduled to be fixed, but rather it will be removed from Cura.
I'll mark this as a duplicate report but I'll leave it open.  The Cura team will certainly have fresher information regarding what is going on with Wire Printing.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic