Ultimaker/Cura #15662: Model dropping to bed

KireanR, 2023-06-02:
### Application Version

5.3.0 and above

### Operating System

Windows 11

### Printer

Elgoo Neptune 3 Plus

### Reproduction steps

1. Deselect drop to build plate
2. Raise the model up by the desired amount
3. Enable supports and click slice

### Actual results

Slicer drops print to the bed ignoring the settings

### Expected results

Print should stay off the build plate the requested amount.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This happens with any file regardless of the amount that it is lifted off the build plate. I have seen some say that it is do to the no empty layers session in the mesh settings but while I can see that setting in the settings visibility menu when I select for it to be visible it does not show in 5.3 or above

GregValiant, 2023-06-02:
Thanks for the report.
This is the 4th or 5th report of this.

- Turn off "Generate Supports"
- Go to Mesh Fixes and turn off "Remove Empty First Layers"
- Go back and turn on "Generate Supports"

The "Remove Empty First Layers" setting gets hidden when "Generate Supports is enabled but it is still affecting the slice.  There is no mention of the behavior in the ToolTip either.
The Cura team is aware of this problem so I'll mark this as a duplicate of #14989 and go ahead and close this one.  I hope you understand.

GregValiant, 2023-06-02:
Closed as Completed.

closed 2023-06-02 by GregValiant