Ultimaker/Cura #15665: Model failed to slice

fly3110, 2023-06-02:
### Cura Version

5.3.1

### Operating System

Windows 11

### Printer

Ender 3

### Describe your model

Soap dish

### Add your .zip here ⬇️

[failed.zip](https://github.com/Ultimaker/Cura/files/11638691/failed.zip)

GregValiant, 2023-06-02:
Thanks for the report.
I see you moved the model from the midpoint of the bed.  It sliced for me so I can't duplicate the problem.
Try rotating the model or moving it someplace else.  This bug actually seems pretty easy to fool.  Sometimes changing the Initial Layer Height just a little bit works to fool it as well.
This behavior has been reported many times so I'll mark this as a duplicate and go ahead and close this one.  I hope you understand.
[image]

closed 2023-06-02 by GregValiant

fly3110, 2023-06-02:
Thank you. The model won't slice with any settings except the "Standard quality". 
I will experiment more and will let you know what I discover.