Ultimaker/Cura #15667: G-code failure

simonzaggio, 2023-06-02:
### Cura Version

5.3.0

### Operating System

Windows 10

### Printer

Artillery Sidewinder X2

### Name abnormal settings

No, I only set crucial settings to remove elephant foot effect, and distances between support

### Describe model location

Yes

### Describe your model

I successfully sliced the model in Cura, then after 12 hours of printing the fault showed up.

### Add your .zip here ⬇️

Hello, while printing last layer of the model my Artillery Sidewinder X2 started making some weird noises. 
It turned out that it was caused by gcode fault and hotend stared to draw weird "worms" and then tried to decrease the Z level from desired 100.2mm to 10.2mm form unknown reason for me. Hopefully, I destroyed only silicone hotend coat and some part of my 3D model. It was last layer of 3D print. Additionally, I had similar situation and hotend started to scratch my 3D printer bed, but I reacted quickly and turned off the printer.
This bug can be very dangerous for 3D printer as it could give it instructions that can cause self destruction.
I very like Cura, it is very comfortable and simple to use, but after last incident I am afraid to use it.
Down below I added .gcode file and .stl model file 

Greetings for Cura support :DD
[krratownica.zip](https://github.com/Ultimaker/Cura/files/11639166/krratownica.zip)

GregValiant, 2023-06-02:
Thanks for the report.
The gcode file appears to be corrupt and consists of two different gcode files.
This is in the top layer and at this point the Z was 100.0
G1 X115.948 Y182.8 E3401.82282
G1 X115.402 Y182.385 E3401.84791
G1 X115.395 Y182.368 **E3401.84858**
G1 X115.011 Y182.18.51911
G1 X106.054 Y150.456 **E1568.52709**
G1 X106.267 Y150.505 E1568.53508
G1 X106.613 Y150.712 E1568.54983
That's a 2000mm retraction in the middle of a print.  Cura certainly didn't do that.
This is a few lines later and it's a Zhop move so it's from Z100 to Z10.2
G1 F1500 E1567.57968
G1 F300 Z10.2
G0 F9000 X120.321 Y164.891 Z10.2
G1 F300 Z10

And here is how the file ends in the middle of a command line.  No Ending Gcode at all, just a truncated line.
[image]

Is the gcode file you posted from the SD card?  If it was I would say you need to format the SD card.
I know you have a mess there and it is in the gcode.  But what I see in the gcode is exactly what happens when there is a bad memory sector on the SD card.  A file gets written and ends in the bad sector so there is no "end of file" marker.  Another file starts in the bad sector so there is no "Start of File" marker.  The printer doesn't know and so merrily goes from one file to the next without any idea that it's a mess.

I'll leave this open so you can respond but at this point I don't see this as a Cura bug since it is obvious (to me) that you have two files that got intertwined.

simonzaggio, 2023-06-03:
Yes, gcode file is from my pendrive, after that event I formatted it. I had 5 various gcode's on it, so as you mentioned, some of this files got interwined with my desired gcode. 
Everything is clear to me. I will have to format my memory drives more often and have only one gcode file on it.
Thank you for your support!

GregValiant, 2023-06-03:
Thanks for getting back.  I'll go ahead and close this then.  Glad you got it sorted out.

I print from the SD card but I monitor the prints through the USB via a piece of software I wrote.  Whenever the printer tells me it is "Unable to Write to File BIN" (the power-loss back up file) I know it's time to reformat the card.  That is the only indication that a sector on the card has become corrupt.

closed 2023-06-03 by GregValiant