Ultimaker/Cura #15669: Failed slicing: Collector can be slice horizontally but not vertically

lapociampi, 2023-06-03:
### Cura Version

5.3.1

### Operating System

Windows 10

### Printer

FLSUN Super Racer

### Describe model location

The slice is bugged when the model is rotated by 90 degrees with the cylindric part touching the buildplate

### Describe your model

Sliced another version before. I changed some quotas and rounded some angles

### Add your .zip here ⬇️

[FSR_Collettore torretta v3.zip](https://github.com/Ultimaker/Cura/files/11642484/FSR_Collettore.torretta.v3.zip)

lapociampi, 2023-06-03:
I think the problem could be the internal part where the two collectors join. If i tilt the model it creates a support inside the tube, maybe if the print is vertical the slicer doesn't know how to sopport that point.

lapociampi, 2023-06-03:
I have also the previous model that is not bugged:
[Collettore torretta v1.zip](https://github.com/Ultimaker/Cura/files/11642509/Collettore.torretta.v1.zip)

GregValiant, 2023-06-03:
Thanks for the report and the files.
This is an often reported bug that can keep a model from slicing when it is at certain points on the build plate.  The mid-point seems to be the main problem.
Here I've moved your original file off to the side in the X and Y and it sliced.  Sometimes rotating a model will fool the bug and sometimes changing the Initial Layer Height by a little bit (like .20 to .22) will fool the bug.
The Cura team is aware of this bug so I'll mark this as a duplicate.  This should be fixed in 5.4.
I'll go ahead and close this as the Cura team has a lot of examples of this behavior.
[image]

closed 2023-06-03 by GregValiant

lapociampi, 2023-06-03:
Thanks a lot, indeed i tried moving the model but not rotating. Have a nice day :)