Ultimaker/Cura #15672: Slice failed depending on object orientation.

ophirf15, 2023-06-04:
### Cura Version

5.3.1

### Operating System

Windows 11 on Dell XPS15 9500

### Printer

Anycubic Kobra

### Name abnormal settings

No changes to settings from my normal print settings

### Describe model location

Model slices fine when the wheel is vertical. Fails when it is flat

### Describe your model

Model is a wheel for a 1969 Mustang model car Sliced fine before I added a screw hole to it

### Add your .zip here ⬇️

[AK_Mustang (4)_fixed.zip](https://github.com/Ultimaker/Cura/files/11644548/AK_Mustang.4._fixed.zip)

GregValiant, 2023-06-04:
Thanks for the report.
It wouldn't slice for me either.  I exported the model and repaired it with MS 3D Builder and it sliced.  So this might well be a model problem.
[image]

There has been an often reported slicing bug that might be the problem here.  I'm going to mark this as a duplicate.  Since that bug has been worked on and should be fixed in 5.4 I'm going to go ahead and close this.  I hope you understand.

Enough of that.
I had a 1970 Mach I with a 351.  It was a beautiful car.  Red with Cragars all around and headers dumping out side pipes.  Very nice.  Other people thought so too.  That car was stolen 4 times while I owned it.  The last time it got totalled in a wreck.

Here is the repaired file.
[GV_Mustang.zip](https://github.com/Ultimaker/Cura/files/11644614/GV_Mustang.zip)

closed 2023-06-04 by GregValiant

ophirf15, 2023-06-04:
Thanks for sharing the Fixed file! I was later able to slice it by changing the scale slightly (rounded to 30x30mm) So that might be related to the issue.
I have a 69 Mustang Coupe and finally got a non-fastback model for it so I am very excited to get it printed!