Ultimaker/Cura #15673: {initial_extruder_number} incorrectly set if no adhesion option selected

MusicalNinjaRandInt, 2023-06-04:
### Application Version

5.3.0

### Operating System

Windows 11

### Printer

Bresser T-Rex2 (=Flashforge Creator Pro)

### Reproduction steps

Preconditions:

- a model which only uses one extruder (e.g. `3x3_cali_rl.3mf`)
- printer start gCode including a tool switch to `{initial_extruder_number}` (e.g. `creator_pro_.def.json`)

1. Set Build Plate Adhesion Type = None
2. Slice model

### Actual results

tool switch is consistently set to T1 regardless of which extruder has been selected. I assume that `{initial_extruder_number}` is set to a default value if not overridden by any adhesion options.

Attached `3x3_cal_rl.gcode` shows result of slicing the above 3mf file
Line 29 is `M108 T1 ; Select initial extruder`

### Expected results

Tool switch should be to the extruder which is in use for this print job. (i.e. `{inital_extruder_number}` should be set based upon the extruder used for the first layer)

line 29 should be `M108 T0 ; Select initial extruder`

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

1. Global find & replace is selected - Bresser (rebadged Flashforge) use `M108` for tool selection. See `Creator+Pro.global.cfg`
2. Enabling a skirt resolves the issue and the correct tool is selected via the gCode start script.

All referenced files:
[Cura Issue initial_extruder_number.zip](https://github.com/Ultimaker/Cura/files/11645872/Cura.Issue.initial_extruder_number.zip)

GregValiant, 2023-06-05:
Thanks for the report.
I don't know that printer but UltiMaker doesn't write the definition files for 3rd party printers.  They come from the manufacturer or a member of the community.  In your case the definition file you provided does list the author on line 8 in the file as "author": "MusicalNinja",
That printer definition file was not included with any of my installations of Cura.  Did it come from Flash Forge?

I don't see an UltiMaker Cura problem here.  You can adjust the StartUp gocde so it works with your printer, or contact the people who provided the definition files and ask them for support.
I'm going to remove the bug label as this appears to be a definition problem.  I'll leave this open so you can respond.  I hope you understand.

MusicalNinjaRandInt, 2023-06-05:
Just taken a look at where this is replacement is made:

1. `StartSliceJob._buildGlobalSettingsMessage` [Line 486](https://github.com/Ultimaker/Cura/blob/ee0e55cf970e6bdebe839f0e1067b6a4b2884915/plugins/CuraEngineBackend/StartSliceJob.py#L486) calls `ExtruderManager.getInitialExtruderNr`
2. given no adhesion [ExtruderManager Line 328](https://github.com/Ultimaker/Cura/blob/ee0e55cf970e6bdebe839f0e1067b6a4b2884915/cura/Settings/ExtruderManager.py#L328) then calls `getUsedExtruderStacks[0].getProperty("extruder_nr", "value")` located at [ExtruderManager Line 198](https://github.com/Ultimaker/Cura/blob/ee0e55cf970e6bdebe839f0e1067b6a4b2884915/cura/Settings/ExtruderManager.py#L198)

So it would appear that something is causing Extruder 1 to be included in the returned list.

I'm not sure whether the issue is with a specific claiming to use this extruder, with a node that doesn't have an active extruder returning a default from [Line 233](https://github.com/Ultimaker/Cura/blob/ee0e55cf970e6bdebe839f0e1067b6a4b2884915/cura/Settings/ExtruderManager.py#L233) or in the later population of `used_extruder_stack_ids`. My guess would be that there is a weird node created earlier in the process ...

GregValiant, 2023-06-05:
Nice investigation.  Using the additional info I can duplicate this.
It does appear that this is a Cura problem.

I'll put the bug label back onto this and the Cura team will take a look.