Ultimaker/Cura #15674: Slicing Crash

xwingfighter94, 2023-06-04:
### Cura Version

5.3.1

### Operating System

MacOS Ventura 13.2.1

### Printer

Prusa

### Name abnormal settings

no, the right part does not crash

### Describe model location

no

### Describe your model

plane part

### Add your .zip here ⬇️

[AV_N_HS1-left-LW_profile3.3mf.zip](https://github.com/Ultimaker/Cura/files/11645943/AV_N_HS1-left-LW_profile3.3mf.zip)

xwingfighter94, 2023-06-04:
thanks a lot for your help

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic

xwingfighter94, 2024-04-07:
Thanks a lot for you answer and time.
I will try to slice

Franck