Ultimaker/Cura #15675: Lift Head Speed

Marcel734, 2023-06-04:
### Application Version

5.2.1

### Operating System

Windows 11

### Printer

Custom

### Reproduction steps

1. I set the Travel Speed to 100mm/s
2. I set the Z hop speed to 5mm/s
3. I enabled Lift Head in the cooling Options
4. I started the print -> failed
5. I looked at the G-Code
6. The movement of the Z Move from the Lift Head Delay was executed with 100mm/s -> too much for Z movements
7. Changed the Travel speed to 99mm/s ->the Z movementr of the Lift Head Option was executed with 99mm/s
8. Why is the Lift Head Option using the Travel Speed Parameter? It would be logical to use the Z-Hop-Speed Parameter for this Lift Head delay ("Z-Hop Similar") Feature.

### Actual results

Z Axis skips steps, when using "Lift Head" Setting for Cooling.

### Expected results

Change the used Speed parameter from "Travel Speed" to "Z-Hop speed" for the "Lift Head" Option. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I don't think a file would help. The text explanation is sufficient.

Marcel734, 2023-06-05:
Thats not the case in my Sitiuation:

;MESH:NONMESH
G0 F300 X100.324 Y74.195 Z0.8
G0 F5940 X100.417 Y73.96
G0 X99.53 Y74.008
G0 X98.68 Y74.541
G0 X98.455 Y74.455
;Small layer, adding delay
G1 F1500 E0.6601
**G0 F5940 X98.455 Y74.455 Z3.8**
G0 X117.319 Y81.109
G4 P6789
;TIME_ELAPSED:51.041296

99mm/s is used

[CFFFP_LiftHeadFail.zip](https://github.com/Ultimaker/Cura/files/11649406/CFFFP_LiftHeadFail.zip)

GregValiant, 2023-06-05:
I see that the problem did exist in 5.2.1 but it was fixed in the 5.3.0 stable release.  If you upgrade to (at least) that version of Cura you won't see the problem.
I'll leave the bug label but I'll go ahead and close this as it has been fixed.

closed 2023-06-05 by GregValiant