Ultimaker/Cura #15682: Fatal error on start because after i messed up fdmprinter.def.json

Knallkopf97, 2023-06-06:
### Application Version

5.3.1

### Operating System

Windows 11

### Printer

Anycubic Vyper

### Reproduction steps

1. i tried to set up "[Small feature slowdown settings are now settable per mesh.]" (https://github.com/smartavionics/Cura/blob/2159908377c2b55bf999c38dfaa316f952698a29/resources/definitions/fdmprinter.def.json) on the following file: C:\Program Files\UltiMaker Cura 5.3.1\share\cura\resources\definitions

2. The feature doesn´t work. so i paste the original fdmprinter.def.json back in the folder.

Since then i have this Error.

Things i alredy did:
- reinstall
- deinstall; delete "My Computer/users/[user_name]/AppData/Local/cura"; install again
- update GPU Driver
- set back on a system restore point

### Actual results

After this i have a fatal error after start showing this:

"A fatal error has occured in Cura....

**Error traceback**
Traceback (most recent call last):
  File "cura\BuildVolume.py", line 667, in _onStackChangeTimerFinished
    self.rebuild()
  File "cura\BuildVolume.py", line 565, in rebuild
    self._updateScaleFactor()
  File "cura\BuildVolume.py", line 520, in _updateScaleFactor
    scale_xy = 100.0 / max(100.0, self._global_container_stack.getProperty("material_shrinkage_percentage_xy", "value"))
TypeError: '>' not supported between instances of 'NoneType' and 'float'

**Logs**
Thread 0x000020b8 (most recent call first):
  File "C:\Program Files\UltiMaker Cura 5.3.1\share\cura\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 90 in _updateThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00001270 (most recent call first):
  File "threading.py", line 324 in wait
  File "zeroconf\__init__.py", line 2534 in wait
  File "zeroconf\__init__.py", line 1715 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000051a8 (most recent call first):
  File "threading.py", line 324 in wait
  File "threading.py", line 600 in wait
  File "C:\Program Files\UltiMaker Cura 5.3.1\share\cura\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00000a4c (most recent call first):
  File "zeroconf\__init__.py", line 1369 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x0000209c (most recent call first):
  File "C:\Program Files\UltiMaker Cura 5.3.1\share\cura\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00003a68 (most recent call first):
  File "UM\Backend\Backend.py", line 169 in _storeStderrToLogThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000016b4 (most recent call first):
  File "UM\Backend\Backend.py", line 157 in _storeOutputToLogThread
  File "threading.py", line 946 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00002cc4 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00001130 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00000978 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x0000039c (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00004a74 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00002ccc (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00003554 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x000023e8 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00002da0 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00001b4c (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00001c28 (most recent call first):
  File "cura\Machines\ContainerTree.py", line 181 in run
  File "UM\JobQueue.py", line 133 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00002a14 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM\JobQueue.py", line 98 in _nextJob
  File "UM\JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Current thread 0x00004660 (most recent call first):
  File "cura\CrashHandler.py", line 397 in _logInfoWidget
  File "cura\CrashHandler.py", line 182 in _createDialog
  File "cura\CrashHandler.py", line 86 in __init__
  File "cura_app.py", line 165 in exceptHook
  File "UM\Qt\QtApplication.py", line 416 in exec
  File "cura\CuraApplication.py", line 898 in run
  File "cura_app.py", line 239 in <module>

### Expected results

I think i messed up some other files with this. But i dont know which one :/

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2023-06-06:
Thanks for the report.
@smartavionics supports his fork on his Github page.
You deleted the "...appdata\cura\ **local** ..." folders but how about the "...appdata\cura\ **roaming** ..." folder?  It also has installation information in it.
It is scary business to make changes to "fdmprinter.def.json".  Simply changing a switch (like making something "settable per mesh") might enable a check box but it doesn't automatically add code within the program that will accomplish what you are trying to do.
This doesn't really appear to be a bug but I'll leave the label on for now and leave this open.  If you would confirm which fork of Cura you are using it may be helpful.

Knallkopf97, 2023-06-06:
ahh shit it worked. :D
you are a wizard. Thanks ^~^
i never touch that code again xD

closed 2023-06-06 by Knallkopf97