Ultimaker/Cura #15684: cant slice a file previously sliced and printed.

tikitock, 2023-06-06:
### Cura Version

530-alpha+xmas

### Operating System

w11

### Printer

cr10 mini

### Name abnormal settings

no

### Describe model location

no

### Describe your model

sliced and printed previously, several times, file was processed with fusion360

### Add your .zip here ⬇️

[CCR10M_SmartphoneHolder.zip](https://github.com/Ultimaker/Cura/files/11667000/CCR10M_SmartphoneHolder.zip)

GregValiant, 2023-06-07:
Thanks for the report and the file.
After turning off "One-At-A-Time" it sliced for me.
Part of the problem is that you have 30 smart phone holders in the workspace.  Most are .06 x .01 x .04 in size.  There is only one that appears full size.  You have Bridge settings turned on and they won't help with this model because there are no bridges, but 4 cantilevered areas that need support.
This appears to be a problem with the way you have set up this slice rather than a Cura bug.  I'm going to remove the bug label but I'll leave this open so you can respond.
[image]

GregValiant, 2024-06-22:
I'll close this today.

closed 2024-06-22 by GregValiant