Ultimaker/Cura #15686: Crash when export button clicked

sulincix, 2023-06-06:
### Application Version

5.3.1

### Operating System

Gentoo GNU/Linux

### Printer

Ender 3 V2

### Reproduction steps

1. install gentoo openrc gnome
2. download cura
3. executable

### Actual results

Error output:
```
PyInstaller/loader/pyimod02_importers.py:352: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Gtk-Message: 20:09:01.959: Failed to load module "canberra-gtk-module"
Gtk-Message: 20:09:01.960: Failed to load module "canberra-gtk-module"
Fontconfig warning: "/etc/fonts/conf.avail/05-reset-dirs-sample.conf", line 6: unknown element "reset-dirs"
Cyclic dependency detected between "file:///tmp/.mount_UltiMac6D1nu/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltiMac6D1nu/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext
file:///tmp/.mount_UltiMac6D1nu/share/cura/resources/qml/PrintSetupSelector/Recommended/RecommendedPrintSetup.qml:96:17: QML QQuickItem: Binding loop detected for property "height"
file:///tmp/.mount_UltiMac6D1nu/share/cura/resources/qml/PrintSetupSelector/Recommended/RecommendedPrintSetup.qml:96:17: QML QQuickItem: Binding loop detected for property "height"

(UltiMaker-Cura:30159): Gtk-WARNING **: 20:09:11.885: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Adwaita-old/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Adwaita-old/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
/tmp/.mount_UltiMac6D1nu/AppRun: line 25: 30159 Aborted                 $scriptdir/UltiMaker-Cura "$@"
```

note: modern.appimage not crashed but do nothing.

### Expected results

...

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

...

sulincix, 2023-06-06:
I found fix:
1. run `./cura.appimage --appimage-extract` command for extract appimage
2. remove libgdk-3.0.so and libgdk-pixbuf-2.0.so files
3. create new appimage file with appimagekit (or directly r AppRun file)

I think gdk-pixbuf loader version and appimage version are incompatible.

MariMakes, 2023-09-04:
Hey @sulincix,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

This issue is very much outside my field of expertise 😦😖
I'll bring it up to the team to see what they can do to improve it.

Fingers crossed 🤞

gcoppex, 2023-10-18:
Thanks @sulincix for your fix. It worked on my end.

@leoheck wrote a similar script (https://github.com/Ultimaker/Cura/issues/11856) that could be adapted for this quick fix:

```
AppImage=./UltiMaker-Cura-5.4.0-linux.AppImage
./${AppImage} --appimage-extract 
mv squashfs-root/libgdk-3.0.so squashfs-root/libgdk-3.0.so.removed
mv squashfs-root/libgdk-pixbuf-2.0.so squashfs-root/libgdk-pixbuf-2.0.so.removed
./appimagetool-x86_64.AppImage squashfs-root 
mv -f Ultimaker_Cura-x86_64.AppImage ${AppImage}
```
As @leoheck noted, this script requires [appimagetool](https://appimage.github.io/appimagetool/)

MariMakes, 2023-10-19:
Quick update from our side 👋

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/16932
We have a ticket on our backlog with the intent to improve this behavior.

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/16932

closed 2023-10-19 by MariMakes