Ultimaker/Cura #15687: Won't slice more complicated model with a lot of honey combs in 5.3.1

Crylia, 2023-06-06:
### Application Version

5.3.1

### Operating System

Pop_OS! Linux

### Printer

Ender 3 V2

### Reproduction steps

1. Download the cura appimage
2. Setup as normal
3. Load model and try to slice
4. Stuck at 40%ish and sometimes slices after hours of working (i5 8600k, rtx 3070)

### Actual results

Takes hours to slice the model while it shouldn't

### Expected results

Should slice normally like it does in Cura 5.2

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

Works just fine in Cura <5.3

GregValiant, 2023-06-07:
That's a bad link to the model so I can't do anything with this.

Crylia, 2023-06-07:
@GregValiant Sorry, no idea what Github did there to the link

https://www.thingiverse.com/thing:5437525

Crylia, 2023-06-07:
I'll attach the project file later, I tried it with the base settings so I thought it would be fine without. Are there session specific logs I could create?

GregValiant, 2023-06-07:
The Cura.log file is in the "Configuration" folder.  In Cura you can use the "Help / Show Configuration Folder" to find it.
There is a plugin in the Marketplace for "Extensive Support Logging".  Even with it installed the log file is cryptic.  I'm starting to learn how to muddle through but it's slow going.

Lines that have DEBUG in them are fine.  WARNINGS need to be looked at.  ERROR is a problem.  Sometimes a description may be contained in the lines following the ERROR lines.  
"Backend exited abnormally with code 3247....771" is a memory error I guess.  I don't get that one but a lot of people seem to.  It happens during slicing so it's a definite problem.

Crylia, 2023-06-08:
Here is the log:
[cura.log](https://github.com/Ultimaker/Cura/files/11692217/cura.log)
Here is the project file: 
[storage.zip](https://github.com/Ultimaker/Cura/files/11692228/storage.zip)

Crylia, 2023-06-08:
If the logs are correct then it took about 1h and 30min to complete the slice, while it takes just a few seconds in 5.2.

GregValiant, 2023-06-09:
There is a huge time difference between 4.13.1 and 5.3.1 as well.

I see all the bug reports here and it is quite apparent that 5.3.1 has issues when slicing a model.  Both "Slicing Failed due to an unexpected error", or "takes forever", might be symptoms of that "slicing problem".  I'm sure the Cura team is trying to address the problem for 5.4 as it is the primary issue with 5.3.1.  We'll see how they make out.

MariMakes, 2023-11-16:
Quick update from our side 👋
We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, I stumbled upon your issue and this seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/17104

Currently the workaround is disabling Optimize Wall Printing Order. 
We have a ticket on our backlog with the intent to improve this behavior.

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/17104
Thanks for your report 👍!

closed 2023-11-16 by MariMakes