Ultimaker/Cura #15688: headphone stand

detectivecosplays, 2023-06-07:
### Cura Version

5.3.1

### Operating System

macOS

### Printer

ender 3d pro

### Name abnormal settings

no

### Describe model location

no it do not slice

### Describe your model

it isn't watertight and I tried the mesh tools plugin

### Add your .zip here ⬇️

[headphone stand.3mf.zip](https://github.com/Ultimaker/Cura/files/11672946/headphone.stand.3mf.zip)

GregValiant, 2023-06-07:
Thanks for the report.
When red surfaces are shown on the top of a model it means that the surface is flipped inside-out.  Cura can't figure out how to slice it.
Mesh Tools is pretty good at telling when a model needs help but the repair tools aren't capable enough for your model.  You will need to find a repair site like [This One](https://formware.co/OnlineStlRepair) and fix the model.
[image]
Here is how it looks repaired.  This one slices
[image]

So this is a model error rather than a Cura bug.  I'll remove the bug label but I'll leave this open so you can respond.

GregValiant, 2024-06-22:
I'll close this today.

closed 2024-06-22 by GregValiant