Ultimaker/Cura #15689: Cura 5.4.0 crashes when a file dialog should be shown

strangeattraction, 2023-06-07:
### Application Version

5.4.0

### Operating System

Linux (Manjaro, XFCE)

### Printer

Wimbo SH155

### Reproduction steps

1. Download [UltiMaker-Cura-5.4.0-alpha-cura_10389_148-linux-modern.AppImage](https://storage.googleapis.com/software.ultimaker.com/cura_nightlies/UltiMaker-Cura-5.4.0-alpha%2Bcura_10389_148-linux-modern.AppImage)
2. Run it
3. Press Open file button

### Actual results

Got an error:

>     Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /org/gtk/libgtk/icons/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
>     Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /org/gtk/libgtk/icons/16x16/status/image-missing.png: Unrecognized image file format (gdk-pixbuf-error-quark, 3)
>     Aborted (core dumped)

### Expected results

I expected that File Manager will open

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I checked this reported issue - https://github.com/Ultimaker/Cura/issues/13393
Actually, this is same issue, but it doesn't solved in my case.

qwerty8224, 2023-06-09:
Can I get a link to Kura Flfa 5.4 version for Windows?

strangeattraction, 2023-06-12:
Hi @MariMakes
Could you help here? Or maybe someone else?

MariMakes, 2023-06-15:
Hey @strangeattraction 

Thanks for tagging me. 👍 
This issue is very much outside my field of expertise 😦😖
But I can some of our linux developers to take a look instead. 

Fingers crossed 🤞 

@qwerty8224 the build they are referring to is this nightly version: https://github.com/Ultimaker/Cura/discussions/15098. No beta yet, but I expect that one to be released soon™️

matulek, 2023-07-05:
Fedora 34, happens to me. 5.3.1 worked fine with .AppImage.

MariMakes, 2023-07-25:
Hey all 👋 

So I'm a Windows user but I've been told that you can try to `apt install xdg-desktop-portal-gtk`

You can set the `QT_QPA_PLATFORMTHEME `to one of the following values: 
- gtk2
- gnome
- kde
- qt5ct
- xcb
- xdgdesktopportal (current default).

Or start the Cura AppImage with the argument `-platformtheme ` with one of the following:
- gtk2
- gnome
- kde
- qt5ct
- xcb
- xdgdesktopportal (current default).

There could be more, possible values are scattered over the internet; Feel free to complete this list.
Can you let us know if that works for you? 

If that's not the case please let us know and we'll be happy to take another look.

strangeattraction, 2023-07-27:
> Hey all wave
> 
> So I'm a Windows user but I've been told that you can try to `apt install xdg-desktop-portal-gtk`
> 
> You can set the `QT_QPA_PLATFORMTHEME `to one of the following values:
> 
>     * gtk2
> 
>     * gnome
> 
>     * kde
> 
>     * qt5ct
> 
>     * xcb
> 
>     * xdgdesktopportal (current default).
> 
> 
> Or start the Cura AppImage with the argument `-platformtheme ` with one of the following:
> 
>     * gtk2
> 
>     * gnome
> 
>     * kde
> 
>     * qt5ct
> 
>     * xcb
> 
>     * xdgdesktopportal (current default).
> 
> 
> There could be more, possible values are scattered over the internet; Feel free to complete this list. Can you let us know if that works for you?
> 
> If that's not the case please let us know and we'll be happy to take another look.

Hello
I tried this, but still getting same error

strangeattraction, 2023-07-31:
UPD: [UltiMaker-Cura-5.4.0-linux-modern.AppImage](https://github.com/Ultimaker/Cura/releases/download/5.4.0/UltiMaker-Cura-5.4.0-linux-modern.AppImage) works fine for me, but [UltiMaker-Cura-5.4.0-linux.AppImage](https://github.com/Ultimaker/Cura/releases/download/5.4.0/UltiMaker-Cura-5.4.0-linux.AppImage) still crashes

MariMakes, 2023-08-07:
Hey @strangeattraction,

Thanks for the update 😄 
I'm glad you are not stuck anymore.

I'll close the issue.

closed 2023-08-07 by MariMakes

laelath, 2024-02-26:
I am seeing this issue on the 5.6.0 AppImage on both my Linux machines, both running arch linux one running Sway, the other i3. I also tried the 5.4.0 AppImage and running it with passing the various `-platformtheme` options suggested.

dynamodan, 2026-09-30:
Still seeing this issue on Debian 12 in 5.4.0.

Newer versions segfault without error just after choosing "skip" on the signin screen.