Ultimaker/Cura #15691: Cura Crashes on mac os it attempts to but then shuts off

ConfiguratorX, 2023-06-07:
### Application Version

5.3.1

### Operating System

Mac OS Ventura

### Printer

Elegoo Neptune 3 Pro

### Reproduction steps

i installed cura and was ready to open and set up my settings 

### Actual results

when i tried to open it verification went fine then it just shut off when i tried to open it it just tried to open but then just shut off

### Expected results

verification goes fine then cura boots and lets me use it

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This Is An Intel 2017 mac btw

GregValiant, 2023-06-07:
Thanks for the report.
I know you can't run it, but if you could find the Cura Configuration folder you should find the Cura.Log file.  Zipping that and posting the zip folder here could be really useful.  I'm a Windows user so I don't know what the path would be on a MAC.

ConfiguratorX, 2023-06-10:
> Collaborator
the log file is non existent ive checked almost 35 different sites and looked where they but no cura directory no log file

closed 2023-10-28 by NoahStoessel