Ultimaker/Cura #15694: wire printing failed

wyatt1414, 2023-06-07:
### Cura Version

5.3.1

### Operating System

macOS Catalina

### Printer

ender 3 v2

### Name abnormal settings

wire printing

### Describe model location

no no matter the position

### Describe your model

prints succesfully without wire printing have not tried mesh fix

### Add your .zip here ⬇️

[Oyster_fungus_shelf.3mf.zip](https://github.com/Ultimaker/Cura/files/11680751/Oyster_fungus_shelf.3mf.zip)

GregValiant, 2023-06-07:
Thanks for the report.
Wire printing is broken in 5.3.1 and my understanding is that it is slated to be removed from Cura.
This is a duplicate bug report about the problem.  I'll leave this open though as one of the reasons it is to be removed was a lack of interest but this report is more proof that folks use it.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic