Ultimaker/Cura #15698: Slicing a project results in a series of dots printing before walls are started

TreyWaters, 2023-06-08:
### Application Version

5.3.1

### Operating System

Windows 10 / PC / Ryzen 5 1600 / Radeon RX 580

### Printer

Elegoo Neptune 3 Plus

### Reproduction steps

Repro Steps:
1. Loaded .STL
2. Sliced .STL
3. View Preview

When slicing with Cura downloaded from Ultimaker, each of the layers has a series of dots that prints before the walls are printed. Normally I probably wouldn't have noticed/cared, but my printer crashes on printing these series of dots, so I had to look into what was going on.
When I sliced with the version of Cura Elegoo provides (Cura 4.8.0), these dots are not created, and only the walls print.

Settings for both Cura versions:

Layer Height: 0.2mm
Wall Thickness: 1.2
Top/Bottom Thickness: 1.2
Horiz. Expansion: 0
Supports: None
Adhesion: None

I'll follow up with Elegoo, as I wouldn't think the printer should crash. That said, I would think that these series of dots on the layers would reduce printing efficiency as opposed to just getting straight to printing the walls.

Attaching .gcode files and .stl file from project.

### Actual results

Sliced project prints dots before printing walls.

### Expected results

Sliced project goes straight into printing walls.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Cura Bug Report.zip](https://github.com/Ultimaker/Cura/files/11687709/Cura.Bug.Report.zip)

GregValiant, 2023-06-08:
Thanks for the report and the file.
Looking closely at the preview - the dots are green and so they are very short "Inner Walls".  At every intersection in the grid the outer wall extrusions leave a diamond shaped gap.  Cura is filling that gap.  You have the wall order set to "Inner to Outer" and so those very short inner wall "dots" are going in first.
You can also see in the preview that the dots don't happen until the second layer.  That's because your Initial Layer Line Width keeps them from forming.
If you set your "Line Width" to 0.50 and your Minimum Wall Line Width to 0.45 the gaps will be small enough that Cura won't bother to fill them.
I don't see a bug here.  The way you have the slice set up leaves gaps that need to be filled.  Changing the settings will get rid of the gaps and consequently the dots.  The Cura 5.x version Variable Line Width has different settings than the 4.8 version and they need to be taken into account.  I'll remove the bug label from this as it appears to be a settings issue.  I'll leave it open for a bit so you can respond here if you that that is premature.
[image]

TreyWaters, 2023-06-09:
Thank you, Greg, for looking at this, and I'm sorry for the trouble. I'm good to close this issue.

closed 2023-06-09 by TreyWaters

GregValiant, 2023-06-09:
No trouble at all.  It's how I learn too.