Ultimaker/Cura #15701: Box Top Failure

TheSpiceOfLife, 2023-06-09:
### Cura Version

5.3.1

### Operating System

Windows 10 Pro

### Printer

Ultimaker S3

### Name abnormal settings

None That I know of 

### Describe model location

I have tried flat, 90 and 30 degrees and stil nothing

### Describe your model

It is a box lid

### Add your .zip here ⬇️

[UMS3_CardBoxTop v7.zip](https://github.com/Ultimaker/Cura/files/11702811/UMS3_CardBoxTop.v7.zip)

GregValiant, 2023-06-09:
Looking at this view, it appears that something is going on with the model.  The red areas have been determined to need support.  But it appears to be very inconsistent (note the W and A in (Wars").
[image]

Computers are stupid, but they are extremely accurate.  You have rotated the model so the overhang angle is exactly 60° and you have set the "Support Overhang Angle" to exactly 60°.  That appears to be creating chokepoints in the math calculations.
Changing the "Support Overhang Angle" to 61° makes a huge difference.

This sliced in about 5 seconds whereas with the Support Overhang Angle at 60° it was taking forever.
[image]

I'll leave the bug report label on this and the Cura team will take a look.  This may be a clue as to what is going on with Tree Supports.

GregValiant, 2023-06-13:
I think it's all part and parcel of the same bug.  I don't know how Cura works under-the-hood but it seems like when the slice goes through a model that some combination of features is causing a problem.  If the slice was to always go from say, left to right, then changing the model orientation can make a difference.  It gives the slicing engine a different look.
I've had some luck just changing the Initial Layer Height.  It usually doesn't take much, like from 0.20 to 0.22.  The slicer goes through a slightly different Z to start out and that changes the Z of all the slices.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic