Ultimaker/Cura #15704: Incorrect Printing Temperature

augustus-thomas, 2023-06-09:
### Application Version

5.3.1

### Operating System

MacOS 13.2.1

### Printer

Prusa i3 Mk3/Mk3s

### Reproduction steps

1. Select "Prusa i3 Mk3/Mk3s"
2. Select "Generic PLA"
3. Navigate to print settings dropdown on right
4. Click on "Show Custom" next to "Recommended Print Settings"
5. Scroll down to "Material" subsection
6. See that the "Printing Temperature" (Nozzle temp) value is autofilled as 200 degrees C which disagrees with the 215 degrees C recommended nozzle temperature provided by [the printer's help page on PLA](https://help.prusa3d.com/article/pla_2062).

### Actual results

1. Changing the default temperature to the correct one causes Cura to "recommend" a revert incorrect settings via the blue revert UI icon. This causes user to think 215 is incorrect.
2. This bug in the profile causes generic PLA to not adhere under the "recommended" settings for this printer.

### Expected results

"Printing Temperature" is autofilled as 215 degrees for Prusa i3 Mk3/Mk3s printer and Generic PLA.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

## initial incorrect default profile
[image]

## after correcting
[image]

GregValiant, 2023-06-09:
Thanks for the report.
This is from the definition file for your printer:
        **"author": "Prusa Research",**
and
        **"has_materials": true,**
It would seem to me that they should provide a material that works with their printer, or add a couple of lines to the printer definition file to set the PLA printing temperature.  Prusa chose not to do either of those.
You can make a custom material (using the Generic PLA as a base for it) and on the Print Settings tab change the Default Printing Temperature to 215.

I'll remove the bug label from this as the problem is with the definition file that was provided by Prusa.  I hope you understand.

augustus-thomas, 2023-06-09:
I completely understand you switching from bug to discussion, as it is not an issue on the Cura side of the codebase! It seems that Prusa develops their own GCode slicer called the "PrusaSlicer" so perhaps they are deliberately not providing definitions so as to promote the use of their software. Although PrusaSlicer is open source so this does not make much sense.

In the meantime I will use my custom **215 C** settings as a fix.

GregValiant, 2023-06-09:
_"...so perhaps they are deliberately not providing definitions"_
I am absolutely 100% sure that isn't it.  There are only so many software engineers who work on the slicers and I'm sure there is some go-between at least among the individuals involved if not at the corporate level.
On both sides of the fence these are busy people.