Ultimaker/Cura #15707: Bottom printing issues

Egh2oboy, 2023-06-09:
### Application Version

5.3.1

### Operating System

Windows 11 home

### Printer

Creality CR10s

### Reproduction steps

I updated to 5.3.1 and now having printing issues. It was dialed pretty well with the prior release. But I cant seem to cure this. Images included. This happens on the bottom of all my prints. I even tried replacing the nozzle tip. Any help is greatly appreciated. 
[image]
[image]
[image]
[image]

### Actual results

Seems the prints on the bottom side of things are just strings. It does seem to clear up after about 1/4 printing. And yes ALL my prints

### Expected results

Should be smooth

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

No additional files

GregValiant, 2023-06-11:
Thanks for the report.
There are some settings that can effect that.  Load the model, set Cura up to slice, and use the "File | Save Project" command and zip the 3mf file and post the zip file here.

Egh2oboy, 2023-06-13:
Was hoping for an update. Please advise
Ed

GregValiant, 2023-06-13:
The first link gives me an error.  The second link is a gcode file and only tells me what the model looks like.

Load the model in Cura, set up to slice.  Use the "File | Save Project" command.  Zip the 3mf file it will create and post the zip folder the.  Do not use Email.  If the folder is over 25mb you will have to put it up on a storage site and provide the download link (that you do via email).
A project file contains the model, all your settings, and your printer.