Ultimaker/Cura #15709: Slow down first layer above infill

StiOII, 2023-06-10:
### Is your feature request related to a problem?

On high speeds on this level  lines of filament breaks due to the high speed when the distance between infill lines are reasonably big.

On low speed printing problem is gone.

### Describe the solution you'd like

New Option for this layer printing speed

### Describe alternatives you've considered

not really any

### Affected users and/or printers

All users who have the same problem.

yet-another-average-joe, 2023-06-10:
Same here after I switched to Klipper with Input Shaper...
At some point this layer could be considered as bridging and using bridging parameters.

But couldn't gradual infill solve the issue ? Didn't test as the issue occured yesterday with a low inffill, quick and dirty model (I usually set infill to 40% so I didn't notice until now).

GregValiant, 2023-06-11:
The "ChangeAtZ" post processor could also be used.  It would be for a whole layer rather than just over the support.