Ultimaker/Cura #15711: Slicing failed

effinay, 2023-06-10:
### Cura Version

5.3.1

### Operating System

Windows 10 22H2

### Printer

Creality Ender-3 Pro

### Name abnormal settings

tree supports touching buildplate, 15% infill, brim, .28 layer height

### Describe model location

center, 45 degrees clockwise

### Describe your model

model sliced before. now trying at 116.71%

### Add your .zip here ⬇️

[UpperPart11671-E3P.zip](https://github.com/Ultimaker/Cura/files/11713476/UpperPart11671-E3P.zip)

GregValiant, 2023-06-12:
Thanks for the report and the file.
This took a while but it finally finished slicing for me with your settings.  It's a 27 hour print.
I'll leave the bug label on this as it appears to be related to tree supports.  The Cura team will take a look.

BTW...changing the Support Overhang Angle to 60° knocked 5 hours off the print time.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic