Ultimaker/Cura #15712: Writing two digits number problem

shoxzz, 2023-06-11:
### Application Version

5.3.1

### Operating System

Windows 11

### Printer

Voron 350/cr10/custom all of them

### Reproduction steps

When i try to write a two digit number, this expecially happens in infill , after the first number cura "confirms" the first digit and then i have to click back again in the box and try to write into it again.
This issue can happen also twice in a row, and you need 3 attempts to have the correct number in the box

### Actual results

Cura confirms a non complete value and unselect the box

### Expected results

you should be able to digit the full number and then move on

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This issue happens on multiple computers and different OS w10 and w11

GregValiant, 2023-06-11:
Thanks for the report.
I don't have any problems duplicating this one.  The Cura team will take a look.

shoxzz, 2023-06-14:
maybe this happens only when you have an high amount of setting displayed, idk