Ultimaker/Cura #15713: Slicing Failed Error

Carlomarmontes, 2023-06-11:
### Cura Version

5.3.1

### Operating System

windows 11

### Printer

anycubic S

### Name abnormal settings

yes

### Describe model location

no 

### Describe your model

Veronie structure 

### Add your .zip here ⬇️

[Vernonie_Cube.zip](https://github.com/Ultimaker/Cura/files/11713653/Vernonie_Cube.zip)

GregValiant, 2023-06-11:
Thanks for the report.
It wouldn't slice for me either.  I dropped the Initial Layer Height to .18 and it sliced.
I'll mark this as a duplicate slicing error and the Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic