Ultimaker/Cura #15714: This is a serious bug regarding slicing algorithms.

SuperDeron, 2023-06-11:
### Application Version

5.3.1

### Operating System

windows10-intel11700K-NV3070Ti

### Printer

Voron2.4-R2-350mm

### Reproduction steps

[image]
[image]
[image]
[image]
[image]
[image]
I found this bug when printing this pipeline model, and only returned a query to find that it was caused by cura slicing. The same parameters were not found to be problematic in prusaSlicer, but the problem was that curved pipeline models like this would experience layering issues when reaching a certain height, some appearing externally and some internally, which is related to the volume of the model. The location is approximately 10% of the total height, I only discovered this issue on the surface of the model after printing, which is a serious bug. I hope Cura can take it seriously. Thank you.

### Actual results

A circular step or staggered layer appears in the printed product.

### Expected results

I hope Cura can solve this serious bug as soon as possible.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]
[image]
[image]
[image]
[image]

smartavionics, 2023-06-11:
Set hole horizontal expansion to zero?

GregValiant, 2023-06-11:
That's what it looks like to me.
@SuperDeron this looks like a settings issue.  See if Hole Horizontal Expansion is set to "0" and re-slice.  See if that ledge goes away.
I'll leave this open for new but it looks like expected behavior.  If you would post a project file it would be helpful.

SuperDeron, 2023-06-12:
It is indeed a problem with the holes, but I was negligent. Thank you for your help.

closed 2023-06-12 by SuperDeron