Ultimaker/Cura #15715: Gyro air slicing fails

rozmansi, 2023-06-11:
### Cura Version

5.3.1

### Operating System

macOS Big Sur (tested on Windows first)

### Printer

Ultimaker S5

### Name abnormal settings

Tried various infill algorithms, wall thickness, profiles - no avail

### Describe model location

Fails whatever orientation or position I try.

### Describe your model

It's the Ultimaker's gyro-orb as shown on Ultimaker's PVA demo downloaded from Nils_R_@Thingverse and unmodified.

### Add your .zip here ⬇️

[UMS5_Gyro_the_Dodo_by_Virtox_-_CC-BY-ND-NC_-_Ultimaker_Special.3mf.zip](https://github.com/Ultimaker/Cura/files/11714037/UMS5_Gyro_the_Dodo_by_Virtox_-_CC-BY-ND-NC_-_Ultimaker_Special.3mf.zip)

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic

rozmansi, 2024-04-05:
I am confirming this slices successfully now. Thanks. 💐