Ultimaker/Cura #15717: Gyro from Ultimaker PVA video

Chichnillasrcute, 2023-06-11:
### Cura Version

5.3.1

### Operating System

Windows server 2019

### Printer

Ultimaker S5

### Name abnormal settings

Supports should work, as the were in the video, the only changes were in making wall thickness equal and sizing it up

### Describe model location

No

### Describe your model

I have but it was a black ABS. i need it as green PLA

### Add your .zip here ⬇️

https://we.tl/t-QSOZcp5Kya
(project file)

fieldOfView, 2023-06-11:
I think you may have forgotten to say what your problem is. We need way more information to help you.

rozmansi, 2023-06-11:
This is a duplicate of #15715. Reported twice due to bad family communication, sorry.

Chichnillasrcute, 2023-06-11:
Still maybe try slicing and posting a .ufp, maybe

Chichnillasrcute, 2023-06-11:
never mind this issue.  post to #15715

closed 2023-06-11 by Chichnillasrcute