Ultimaker/Cura #15718: Slicing failed

kr320, 2023-06-11:
### Cura Version

5.3.1

### Operating System

Windows 10

### Printer

Ender 3 s1 pro

### Name abnormal settings

Scaled to 50% and then error occured

### Describe model location

Rotated 45 degrees and it sliced, rotated back to previous location and it now slices

### Describe your model

Yes, when scaled at 100% and 70%. Yes watertight. Did not try mesh fixes since I got it to slice correctly

### Add your .zip here ⬇️

[CE3S1PRO_Flexi_Hammerhead_shark_V2-20230510.zip](https://github.com/Ultimaker/Cura/files/11715142/CE3S1PRO_Flexi_Hammerhead_shark_V2-20230510.zip)

GregValiant, 2023-06-11:
Thanks for the report and the file.
I didn't have any trouble slicing it.  Try moving and/or rotating the model on the Cura build plate.  Sometimes making a small change in the Initial Layer Height (like .20 to .22) can fool this bug.
This behavior has been reported numerous times.  I'll mark this as a duplicate and go ahead and close it.  I hopen you understand.
[image]

closed 2023-06-11 by GregValiant

kr320, 2023-06-11:
Very good, always try to help when I can. FYI after I rotated it 45 degrees
it sliced and continued to slice fine. Have a good one!