Ultimaker/Cura #15719: MacOS Sonoma

errorexists, 2023-06-11:
### Application Version

5.3.1

### Operating System

MacOS Sonoma

### Printer

Max Neo

### Reproduction steps

can not reopen after minimized have to right click on app in dock select Show all windows to reopen app   

### Actual results

then opens

### Expected results

open an minimize should be working 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

MariMakes, 2023-07-17:
Hey @errorexists,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

What an interesting report 🤔 
I'm a Windows user myself, but this shouldn't happen. 

Can you try to install the PKG version of Cura?
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.4.0

Can you let us know if this resolves your issue?
If it does not I'll bring it up to the team so they can take a look.

Thank you 🙏 
Thanks again, and please let us know if you run into any other issues 💪

errorexists, 2023-07-17:
hello @MariMakes  no worries an thanks for your reply
im thinking its a os issue as with mac os sonoma as since this last post im now with a newer version an now it shows in the dock but clicking on the icon doesnt open it i still have to right click an im on the latest version of cura 
an also in the pic youll see a translucent box with the arrow pointing to it not sure what that is either 
im on Mac os sonoma beta 14.0 Beta (23A5286i)
thank you 
[image]

oleksii-suprun, 2023-10-08:
Recently I've updated to the latest MacOS Sonoma. I don't have any slicing issues with Cura 5.4, but I can confirm that when I open Cura, an additional small minimized window appears.

[image]
[image]
[image]
[image]

DavidAppleWWDR, 2023-11-02:
An Apple AppKit Engineer looked into this for you! Passing it on on their behalf:

> **Problem**
> UltiMaker Cura calls into AppKit before [NSApplication run] with no autorelease pool, which leaks objects.
> **Analysis**
> Calling into AppKit without an autorelease pool is fragile because implementation changes inside AppKit may leak new objects. One such object is a window which is miniaturized and then deallocated. A change in AppKit will retain+autorelease this window, which effectively leaks it due to the autorelease pool which is not drained. The leak leaves the window on screen, whereas before it would deallocate and disappear.
> **Resolution**
> UltiMaker Cura should have an autorelease pool which is periodically drains around any calls to AppKit. See problematic backtrace below.
> Additionally, UltiMaker Cura should investigate what this window created at app launch is, and why it’s miniaturized right before being deallocated. Why create the window at all if it’s deallocated a moment later?

foxymichelle, 2024-04-10:
Just checking in to see if there's any movement on this - Sonoma hasn't been in beta for about 6 months now and Apple AppKit Engineer diagnosis posted in November.  Seems like it's time to take this out of the deferred status.

oggreeb, 2024-09-26:
Confirming this is still a problem on MacOS Sequoia (15.0) and Cura (Apple Silicon 5.80 version installed from Github.

Arsoth, 2025-01-13:
Still an issue with Cura 5.9 installed from the Ultimaker website on Sequoia 15.2.