Ultimaker/Cura #15721: Cura blocking Lightburn from using COM port

tbreefing, 2023-06-11:
### Application Version

5.3.1

### Operating System

Windows 10 Pro

### Printer

Kingroon K3PS

### Reproduction steps

Is there a setting in Cura so it doesn't want to automatically connect to what's plugged into the com port? I run Lightburn for my laser and occasionally will run Cura at the the same time but if I open Cura I'll lose communication to my laser. Doesn't matter if Lightburn is open first or if I select the laser in the com list through lightburn to try and get it to reconnect. For some reason if Cura is open Lightburn just won't communicate with the laser. As soon as I close Cura Lightburn will start working again. Looked through the settings but didn't notice anything in regards to this.

### Actual results

If I close Cura I can control my laser through Lightburn again

### Expected results

Cura not automatically connect to something connected to the COM port or at least have a way to disable that option. Of course there very well could be a way already but I've been unable to find it so far.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No files or screenshots as I'm not sure what would be needed to view that type of issue.

GregValiant, 2023-06-11:
Thanks for the report.
Yes, "USB Printing" is turned on by default and grabs the port when Cura starts.
Go to the MarketPlace (icon at the right top of the workspace).  Click on the Gear icon.  Scroll down and disable "USB Printing".
[image]

This is expected behavior so I'll remove the bug label.  I'll leave this open so you can respond here.

tbreefing, 2023-06-12:
Thank you! Would have never known there were settings options in the Marketplace. Appreciate the help!

closed 2023-06-12 by tbreefing