Ultimaker/Cura #15724: HELP, slicing failed for my architecture model

douvine, 2023-06-12:
### Cura Version

5.3.1.

### Operating System

windows 11

### Printer

ender 3 v2 neo

### Describe your model

this is a piece of my architecture model, all of the other part ( 14 of it) turned out fine but this one cant be sliced

### Add your .zip here ⬇️

[CE3E3V2_3D PRINT 02-02.zip](https://github.com/Ultimaker/Cura/files/11716845/CE3E3V2_3D.PRINT.02-02.zip)

GregValiant, 2023-06-12:
Thanks for the report and the file.
This report is from formview.co/OnLineSTLRepair.  The model has some issues.
-> Reading file and indexing vertices
-> Analysed your file:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
**--> 27283 Non-manifold edges (?)
--> 186979 Inverted faces (?)
--> 101004 Degenerate faces (?)
--> 19955 Duplicate faces (?)**
--> 0 Disjoint shells (?)
That repair site has a 4 minute time out and it wouldn't repair the model.  I moved the model on the build plate and it sliced.  I don't see any way for it to be a successful print though.
The orientation of the model on the build plate should not matter.  The printer is accurate enough that it will print the same either way.
I'll leave the bug label on this but the model has so many errors in it that the problem could well be the model.

This is in 5.3.1
[image]

Here is the car.  It has a lot of errors and the results are not good.
[image]

douvine, 2023-06-12:
Hi Greg I've tried to repair my model using online repair tool and then i simplified the face using 3d building now it works fine, thanks for your input. I'm new to this hobby do you have any tips to convert .skp file/.pln file to .stl so that there wont be as much error as it is now? thanks for your respond highly appreciated

douvine, 2023-06-12:
ok thanks for your input ill put it in mind the next time i make architectural models.

closed 2023-06-12 by douvine