Ultimaker/Cura #15728: No Option to Change Nozzle Size

ariberman, 2023-06-12:
### Application Version

5.3.1

### Operating System

Mac OS BigSur

### Printer

CREALITY Ender 3 S1 Pro

### Reproduction steps

Opened an STL file.
Customized and set filament to Tritan.
Selected nozzle size pull down... only 0.4mm available.  Want to run different size (0.8).

### Actual results

Have no alternate nozzle sizes to pick.

### Expected results

List of nozzle sizes or input box for entering a nozzle size.
Or.... place to customize the printer as a new printer with a different nozzle size

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none

GregValiant, 2023-06-13:
This is a Creality problem.  Ultimaker doesn't provide the files for other manufacturers printers.
If you go to the ".../share/cura/resources/variants" folder you will come across a file for your nozzle "creality_ender3s1pro_0.4.inst.cfg".  You can open the file in a text editor, do a save-as just changing the "0.4" to "0.6" or whatever and and change the nozzle size in the file.  Do that for whatever nozzle sizes you want and they should show up in Cura _**IF**_ your printer was configured to have variants.
You can also change nozzles using the box on the Machine Settings Extruder tab.
I'll remove the bug label and close this as it's a Creality definition and files problem.  I hope you understand.

ariberman, 2023-06-13:
Thanks for that info.  
By the way, I've looked for the file "creality_ender3s1pro_0.4.inst.cfg" and can't see to locate the path you mentioned (root/share/cura/resources/variants/creality_ender3s1pro_0.4.inst.cfg).  Is that there a different location on MacOS?

GregValiant, 2023-06-13:
Yeah, I think it's different and I don't know MAC's.  It should be within the installation folder though.
There is also a Configuration Folder ("Help | Show Configuration Folder") and it might be in there someplace.

@MariMakes do you have the path to the variants folder?

MariMakes, 2023-06-20:
Hey @ariberman,

I'm not a Mac user myself but I've troubleshooted something similar in the past. 
Back than we were able to find it here:

1. [Open Terminal](https://www.howtogeek.com/682770/how-to-open-the-terminal-on-a-mac/)
2. Type or copy paste
` open ~/Library/Application\ Support/cura`
4. Open 5.3 folder
5. Open the Variants folder

It that's not it, please let us know and we'll find an actual mac user to help you troubleshoot. 😉

ariberman, 2023-06-20:
>    3. Open 5.3 folder
>    4. Open the Variants folder

Hi Mariska,
Thanks... That finds the variants directories from 5.0 to 5.3, but
the  directories in all four are empty.  The only CFG file in the area is
cura.cfg and that is in the cura directory above variants.
So.... still searching...

casperlamboo, 2023-06-21:
As this configuration file is shipped with you cannot fined it in the shared folder indeed. It is located in the app data instead. To navigate to the correct file follow the following steps.

1. right-click on `UltiMaker Cura.app` in you applications folder and click "Show Package Contents"
2. navigate to `../Contents/Resources/share/cura/resources/variantes/creality/`
3. Open `creality_ender3s1pro_0.4.inst.cfg` in an editor and apply the changes as mentioned by @GregValiant

ariberman, 2023-06-29:
Sorry for the late reply - I'm in the middle of a move.
Greg, thanks for that info.  I'll try it out as soon as I get everything
reassembled here.

Radmcg, 2023-07-02:
Hi Guys new to Github.  Found this thread searching for the same problem.  I had found this information in different forms other places.  Added a 0.6 mm config file as above and the s1 pro profile will still not find it.  I am using a Mac BTW.  I can set up a profile using the plus and I can get all the nozzle sizes.  Not sure why the pro will not find the new 0.6 config file?  AHA appreciated

MariMakes, 2023-07-10:
Hey @Radmcg,

Cura needs the files to meet very very specific conditions, a typo in the file name or file itself will already result in it not being found. I've noticed that deleting the file and trying again might correct any mistakes that I make during the process. If that still doesn't help I suggest to reach out to [our community here](https://community.ultimaker.com/) instead of our issue tracker.