Ultimaker/Cura #15729: Pause at height setting resume temperature to zero in generated GCODE

billgeek, 2023-06-12:
### Application Version

5.2.1; 5.3.1

### Operating System

Windows 11 x64

### Printer

Voron 2 300mm (I named it "Voron!")

### Reproduction steps

1. Load up any model. I'm using "New latch 6.0mm.stl" from here: https://mods.vorondesign.com/detail/9Rdnf5vD2oaJLmR7BpAuQ
2. Select Extensions > Post-Processing > Modify G-Code
3. Click "Add Script"
4. Select "Pause at height"
5. Set the layer number to number 2 (so that the pause happens after layer 2)
6. Set standby temperature to 230C
7. Add some display text (For own reference: "Change to Green")
8. Set G-Code before pause to "PAUSE" (Klipper macro name)
9. Click "Close"

Slice the model and save to disk.

### Actual results

When editing the generated GCODE file I can see the PAUSE inserted, but the resume temperature is always zero:
```
;added code by post processing
;script: PauseAtHeight.py
;current layer: 2
M83 ; switch to relative E values for any needed retraction
G1 F300 Z1.6400000000000001 ; move up a millimeter to get out of the way
G1 F9000 X190 Y190
G1 F300 Z15 ; too close to bed--move to at least 15mm
M104 S230 ; standby temperature
M117 Change to Green
PAUSE
M0 ; Do the actual pause
M104 S0 ; resume temperature
```
Note the last line: M104 S0 - This also happens when using M109 instead.

### Expected results

I expect the printer to:
1. Start the print at 255C as per the "base" settings in Cura
2. Once reaching the end of layer 2, execute the PAUSE command which parks the nozzle and drop the nozzle temp to the desired 230C as configured
3. Once I press resume, I expect the printer to resume at 255C (or perhaps even 230C for parking) but NOT zero degrees.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Note that I have other gcode modifiers included:
1. Generate 32px thumbnail
2. Generate 300px thumbnail
3. Insert `TIMELAPSE_TAKE_FRAME` before each layer change

These are the settings I use for the Pause at height:

[image]

And here's the end-result I'm seeing:

[image]

I've tried playing around with just about every setting in this dialog, I've removed other scripts, etc... I have also reproduced this using 5.2.1 and it doesn't seem to matter which printer I use. (I also have an Ender 3 profile where this also happens)

I do recall this working in the past, so not sure if this is something I'm doing that accidentally resets the temperatures, or if this is genuinely a bug.

Any help would be appreciated!

Attached below is my project file in ZIP format:

[CuraPauseIssue.zip](https://github.com/Ultimaker/Cura/files/11725537/CuraPauseIssue.zip)

GregValiant, 2023-06-15:
Thanks for the report and the file.
I think I can see what is going on but your 3mf file opened with a single extruder printer."Voron!".  Because of that I can't actually duplicate the problem.  If you could post that gcode file it may help.

The PauseAtHeight script follows the temperature through the file so it knows the resume temperature.  It's possible that what happened in your case is that the pause was inserted immediately after the last use of T1 and Cura had shut that hot end down.
If the last temperature line before the pause was "M104 S0 T1" it would appear that the way the code is written that the script would find both the ActiveExtruder and the ResumeTemperature in that line.  They would both be wrong.

So this looks like a bug to me as well.  Maybe if @rburema gets a minute he can take a look at this?

billgeek, 2023-06-15:
Thanks for the feedback. Attached is the generated GCODE file from v5.3.1 as requested:
[ReproduceIssue531.zip](https://github.com/Ultimaker/Cura/files/11761665/ReproduceIssue531.zip)

I did play around a bit more and it appears as though the order of post-processing scripts could be the issue. 
Moving the "Insert at layer change" script below "Pause at height" produces an accurate file.

1. Load up a model
2. Add post-processing script "Insert at layer change"
3. Add post-processing script "Pause at Height" (Confirm that the order is FIRST insert at layer change, then pause at height)
5. Slice, save to disk and open the file in a text editor
6. You will notice the resume temperature being zero

[image]

1. Move "Insert at layer change" below "Pause at height"
2. Slice, save to disk and open the file in a text editor
3. Notice the resume temperature is accurate

[image]

rburema, 2023-06-15:
Thanks for the report! (Also hi Greg!)

Looking through the supplied project-file, I came to much the same conclusion (that'll teach me to read the entire discussion before diving in...):

The first 'T' in 'TIMELAPSE_TAKE_FRAME' inserted by the previous script, is interpreted by the pause-at-height script as an extruder switch.

Since no default is given in `setValue` when getting the extruder-number from the 'T' command, so it gets interpreted as `None` instead of a sensible number like `0`. It then looks for the most recent known temperature for the `None` extruder, which of course it can't find, since no M104 or M109 commands are ever given for the `None` extruder. However, a default value of `0` _is_ given for when it can't find the temperature of the extruder, so that gets written.

The permanent fix would be to have the default value for the extruder be the previous (or rather current) one.

closed 2023-06-15 by rburema

GregValiant, 2023-06-15:
Line 341 in the PauseAtHeight script is:
_**if line.startswith("T")**_ and the script is evaluating your "TIME_LAPSE_..." line as a Gcode command because it starts with a "T".

Really, only Gcode and Mcode commands should be on a line.  Everything else should be preceded by a semi-colon to make it a comment.
I guess I'll leave the bug label on this.  As Cura wasn't written with Klipper in mind then this is an issue.
I have changed this file so it works.  Put in in the Configuration Folder (in Windows it should be "C:\Users\.......\AppData\Roaming\cura\5.3\scripts".  The name will be slightly different so you know which one to use.
[gv_PauseAtHeight.zip](https://github.com/Ultimaker/Cura/files/11763596/gv_PauseAtHeight.zip)

rburema, 2023-06-16:
> Really, only Gcode and Mcode commands should be on a line. Everything else should be preceded by a semi-colon to make it a comment.

I completely agree, ideally. Some firmwares do accept commands like that however, and I'm not sure if they'd still parse properly if commented out :-/

> this what I did.

I just solved it like this: https://github.com/Ultimaker/Cura/commit/99e1fcd4f8200bff2ca03ed8a15f43ddf6ee9d25

In a way your solution is a bit closer to what's actually meant though. However, `re.search(...)` would also find something in `START3B`, or even any gcode commands that (don't start with T but) would have a T parameter with a number. So we'd at least need to replace that with `re.match(...)` (see https://docs.python.org/3/howto/regex.html#match-versus-search)

rburema, 2023-06-16:
@GregValiant I made an extra commit to incorporate your version of the fix instead (but with match instead of search). -- Should show up in Cura 5.5

GregValiant, 2023-06-16:
I agree "re.match" is a better choice.

There can't be many more of those traps are out there.

commit 99e1fcd4 (2023-06-15): Pause-at-height: Under rare circumstances, T might not be temperature.

Some printers accept commands like TIMELAPSE_TAKE_FRAME or whatever.

Should fix #15729
commit fa8e4815 (2023-06-16): Pause-at-height: Under rare circumstances T might not be an extruder switch.

Some printers accept commands like TIMELAPSE_TAKE_FRAME or whatever, after all. Make it explciti that a line should start with T<number> in order to qualify.

Should fix #15729