Ultimaker/Cura #15730: Error slicing

geniusboywonder2020, 2023-06-12:
### Cura Version

5.3.1

### Operating System

Mac OS X Ventura 13.1

### Printer

Ender 3

### Describe your model

Wheel adapter

### Add your .zip here ⬇️

[CE3E3V2_sc2_QR_wheel_side.3mf.zip](https://github.com/Ultimaker/Cura/files/11725754/CE3E3V2_sc2_QR_wheel_side.3mf.zip)

File attatched

geniusboywonder2020, 2023-06-12:
I moved the model just off to the side and it sliced correctly

MariMakes, 2023-07-17:
Hey @geniusboywonder2020,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I'm able to slice your model successfully in Cura 5.4.
Probably because we made a lot of improvements to tree support in the latest version of Cura. 
[image]

It's resolved in the 5.4 stable release, you can download the Cura version with the fix here: https://github.com/Ultimaker/Cura/releases/tag/5.4.0
I'll close this issue since it's resolved. 

Thanks again, and please let us know if you run into any other issues 💪

closed 2023-07-17 by MariMakes