Ultimaker/Cura #15731: AGPLv3 still in comments?

sp7412, 2023-06-12:
### Application Version

5.3.1

### Operating System

Windows

### Printer

Ultimaker

### Reproduction steps

I see that Cura is released under GNU Lesser General Public License v3.0 (https://github.com/Ultimaker/Cura/blob/main/LICENSE) and that the license changed from AGPLv3 to LGPLv3 in 2017 (https://github.com/Ultimaker/Cura/commit/9a193ad5c5ab5324ad1335dd9adf80fef24a7977).

But I still see AGPLv3 in the header of this file https://github.com/Ultimaker/Cura/blob/main/plugins/PostProcessingPlugin/PostProcessingPlugin.py#L2

Was this file missed when Cura went from GPLv3 to LGPLv3 in 2017?

### Actual results

N/A

### Expected results

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

N/A

fieldOfView, 2023-06-13:
The PostProcessing plugin was originally a separate project, released under the AGPL:
https://github.com/nallath/postprocessingplugin

It looks like the plugin was merged into the Cura project (6c0fb110fe842758c7cc20dae7edc055d5c8ad8c) after the change in license on the Cura repository. So that leaves the Cura project with a mixed license at the moment.

sp7412, 2023-06-13:
@nallath

MariMakes, 2023-07-17:
Hey @sp7412,

Thanks for your report 👍
Excellently spotted 🔎 

I'll bring it up to the team to see how they can update it.

fieldOfView, 2023-08-14:
Technically, I was never asked permission to change the license of my contributions _to the Postprocessing plugin repository_ after october 2016. My contributions before that date could be seen as owned by Ultimaker. If you want my permission, then just ask.

nallath, 2023-08-14:
Err. You are right. I overlooked that. Mea culpa! 

So, Do you mind changing it to LGPL?