Ultimaker/Cura #15732: Slicing ender pi enclosure failed slicing with tree support selected

jbhome, 2023-06-12:
### Cura Version

5.1.0

### Operating System

Windows 10

### Printer

Ender 3

### Name abnormal settings

Seleccted tree supports

### Describe model location

As placed by cura except rotated to orient build direction

### Describe your model

Enclosure for raspberry pi for Klipper

### Add your .zip here ⬇️

[CE3PRO_Extended Housing.zip](https://github.com/Ultimaker/Cura/files/11728346/CE3PRO_Extended.Housing.zip)

Sliced OK when normal support selected. Failed when tree support selected.

GregValiant, 2023-06-13:
Thanks for the report and the file.
This is another one that was causing someone grief and I didn't have any trouble slicing.
[image]

This bug has been reported numerous times so I'll mark this as a duplicate.  I'm going to leave this one open for the Cura team though since it involves tree supports.

Did you try rotating the model or moving it around on the build plate in Cura?  Sometimes changing the Initial Layer Height by a little bit can fool this bug as well.  This bug is being worked on and hopefully will be fixed in 5.4.

jbhome, 2023-06-13:
Hi Greg,

I did not try any other options. I did turn off tree supports and resliced
it without problem. First issue I've had with cura. Awesome work.
Now.. if we could get some open source support for Android predictive text.
I am so tired of that drunken clown and his random spelling correction.
Have a great day.

Regards,
Jim Brooks

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic