Ultimaker/Cura #15736: Print with 2 nozzles

GabyChab, 2023-06-13:
### Application Version

5.3.1 / 5.3.0 / 5.0.0

### Operating System

W11 

### Printer

ultimaker S5

### Reproduction steps

1. I choose to do a job with a PLA in first nozzle and PVA in the second, for the support.
2. I put in Cura the right nozzle in the right thing and I slice it.
3. Then I print it
4. All comes  with the same nozzle ( the fisrt one in PLA). It's look like the second nozzle work but don"t have any line in the Gcode.

### Actual results

I thought in first, I made a mistake somewhere. 
So I check everything, I clean all my job and I redo it again. Same result.
After that I did a print with two stl, with differents nozzle for each one and I notice that when I chosing a nozzle inside the parameter it change automaticaly for the two nozzles... So it's impossible to do at the same time two slt with different material ? 
That was working with a previous version of Cura... 

### Expected results

How I can fix this ? 
Did I made an other mistake, and I didn't notice it ? 
Or it's software problem, I that way how it can be improve ? 
Thanks by advance for your help! 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[UMS5_the one _______Filament_Swatch_ReRemix2.txt](https://github.com/Ultimaker/Cura/files/11734602/UMS5_the.one._______Filament_Swatch_ReRemix2.txt)

GregValiant, 2023-06-13:
Thanks for the report and the file.
If you could provide a Project file ("File | Save Project") it would be helpful.
Within the gcode file there are 43 tool changes to T0 and 41 tool changes to T1.  That part appears to be fine.
This is a view of the gcode in Cura.  I have turned off E1 so this is the support extruder.
[image]

Here I have switched and the support extruder is turned off.
[image]

Here is using material color with both extruder shown.
[image]

The gcode looks good.  This may be a printer problem.  I'll leave the bug label on and someone from the Cura team will take a look.

GabyChab, 2023-06-13:
Thanks for your help! 
Here is the save of my project: 
https://www.dropbox.com/s/kgoy7mvfhlapayt/UMS5_the%20one%20_______Filament_Swatch_ReRemix.3mf?dl=0

GregValiant, 2023-06-13:
I sliced your project file and it looks good to me.  The gcode is near identical to the one you came up with.  I don't see a problem there at all.
I think you are going to need one of the UltiMaker folks to take a look at this.

What firmware version is loaded in your printer?  It's something they will need to know.

closed 2023-06-29 by github-actions[bot]