Ultimaker/Cura #15740: portal slice failed

lylmiksistemas, 2023-06-13:
### Cura Version

5.3.1

### Operating System

Windows 11, nvidia gforce gtx 1080

### Printer

FLSUN v400

### Name abnormal settings

None, no not slice with default profiles

### Describe model location

not, not slice, rotate and move and not slice

### Describe your model

no, never slice, mesh tools tells is waterthight, and fix do nothing

### Add your .zip here ⬇️

https://we.tl/t-xQWlE2ETkl this is the project.

The errors are at middle of slicing, fails or start again at 0% and not advance.

thanks.
[FV_Ruined_Archway_base_repaired.zip](https://github.com/Ultimaker/Cura/files/11737580/FV_Ruined_Archway_base_repaired.zip)

GregValiant, 2023-06-13:
Thanks for the report and the files.
I absolutely cannot get this model to slice...with your FLSUN as the active printer.  I switch over to my Ender and it slices in a couple of seconds.
There aren't any errors in the model.  The Cura team will take a look.
Here it is with my printer active.  I went through your settings and didn't see anything odd, it just won't slice.
[image]

lylmiksistemas, 2023-06-14:
Thanks for the reply, and waiting for more news :)

MariMakes, 2023-07-17:
Hey @lylmiksistemas,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Thanks for the Project File 👍
I took a look at it in Cura 5.4 and I was able to slice it succesfully in our latest version of Cura, with your settings. 
That means your issue is resolved 🎉 

[image]

You can download the Cura version with the fix here: https://github.com/Ultimaker/Cura/releases/tag/5.4.0

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. 
Apologies for any inconvenience caused by our delayed response. 
I'll be closing this issue. I hope you understand.

closed 2023-07-17 by MariMakes