Ultimaker/Cura #15743: Support blocker shape change

Scull215, 2023-06-14:
### Is your feature request related to a problem?

I want to create a support/infill pillar for a solid block in only certain areas and also block out supports in threads

### Describe the solution you'd like

I would prefer to use a cylinder instead of the basic cube pattern

### Describe alternatives you've considered

I could use the basic cube form 

### Affected users and/or printers

It should affect everyone even if its minor
My specific file is on a C clamp

GregValiant, 2023-06-15:
Go to the MarketPlace and install the Cylindric Custom Support plugin.
I think that by default when you bring one into Cura is will come in as "Print as Support" but you can change that in the Per Model settings.

5axes, 2023-07-25:
**Custom Support Eraser Plus** plugin in the marketplace
[image]