Ultimaker/Cura #15746: Blobbing on 3D Benchy Printed with Klipper and Cura 5.3 (Resolution Settings)

BigWampa1989, 2023-06-14:
### Application Version

5.3.0

### Operating System

Windows 10 - Klipper

### Printer

Creality Ender 3 

### Reproduction steps

When printing a 3D Benchy on Cura 5.3 using Klipper blobs are formed on the bough of the boat. I realized this was linked to the Resolution settings. When Resolution was decreased from default settings the blobs started to dissapear. While watching the print closely I could see the nozzle slow down at the spots where the blobs were forming. When using Cura 5.2.1 standard resolution formed no blobs. In fact, I increased resolution to extreme levels and still no blobbing occurred on 5.2.1. This is unrelated to anything mechanical, Klipper itself, or other slicer settings. Their appears to be some sort of glitch on 5.3 concerning Klipper use and resolution settings. 

### Actual results

Blobbing on the Bough of the 3D Benchy.

### Expected results

No blobbing on the 3D Benchy. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

The one photo shows a piece cut off the top of the bough, i did that to check to see if the blobs had something to do with the infill. It was not until I reduced the Resolution settings that the blobbing started to decrease. The same and higher resolution setting on Cura 5.2 had no impact on quality and produced no blobbing.

MariMakes, 2023-07-17:
Hey @BigWampa1989,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I recognize this issue. 🔎
This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/14811 where blobs would appear on curved surfaces for klipper printers. 

Luckily it seems to be resolved in the 5.4 stable release (it's a bit hard for us to test), you can download the Cura version with the fix here: https://github.com/Ultimaker/Cura/releases/tag/5.4.0
I'll close this issue since it's a duplicate and resolved. 

Thanks again, and please let us know if you run into any other issues 💪

closed 2023-07-17 by MariMakes

BigWampa1989, 2023-07-18:
Hi Mariska, 5.4 did fix the blobbing issue but their seems to be a new or
related issue concerning the "3DBenchy" text on the rear of the benchy. I
opened a new ticket under Issue # #16207

Best Regards,

*Louis A. Casadia, Esq.*
***@***.***